Codax2000/fir-cnn-rtl #48: Optimizing conv_layer to Use Less logical_units

EugLiu, 2023-05-11:


Codax2000, 2023-05-15:
[image]

Testbench working with 2x5 input, 2x3 kernel

Codax2000, 2023-05-15:
[image]

Block RAM utilization

Codax2000, 2023-05-15:
[image]

CLB and LUT utilization. This will likely be lower on our board, since we have DSP slices.

Codax2000, 2023-05-15:
Synthesis time:
[image]

Codax2000, 2023-05-17:
New post-synthesis simulation report:
[image]
Timing simulation fails hold time (as does synthesis timing report), but there is no clock tree, so wait for post-implementation simulation to sort out hold time violations.

Post-implementation simulation report:
[image]

Timing constraints and implementation (for this testbench at least) are fully functional.

PR #53 (2023-05-17): 48 optimizing conv layer to use less logical units
Passing timing and functional simulations, synthesis and implementation passing successfully.
Resolves #48
closed 2023-05-17 by Codax2000